knipknap/better-tool-library #1: Win 11 (insiders Beta) with FreeCad 0.21.0 (rev 33345) Failure to load

ShamanTcler, 2023-07-25:
09:37:52  Better Tool Library loaded successfully.
09:37:52  Feeds and Speeds Addon loaded: PathWorkbench
09:38:00  Warning: Skipping loading of the following files, because they have reserved names used by a builtin shape: ballend.fcstd, bullnose.fcstd, chamfer.fcstd, drill.fcstd, endmill.fcstd, probe.fcstd, slittingsaw.fcstd, thread-mill.fcstd, v-bit.fcstd --> To remove this warning, delete or rename the files
09:38:00  Traceback (most recent call last):
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\ui\init.py", line 22, in on_library_open_clicked
09:38:00      dialog = LibraryUI(tool_db, serializer)
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\ui\library.py", line 51, in __init__
09:38:00      self.load()
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\ui\library.py", line 101, in load
09:38:00      self.tooldb.deserialize(self.serializer)
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\db.py", line 114, in deserialize
09:38:00      self.deserialize_libraries(serializer)
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\db.py", line 84, in deserialize_libraries
09:38:00      for library in serializer.deserialize_libraries():
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\serializers\fcserializer.py", line 123, in deserialize_libraries
09:38:00      return [self.deserialize_library(id)
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\serializers\fcserializer.py", line 123, in <listcomp>
09:38:00      return [self.deserialize_library(id)
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\serializers\fcserializer.py", line 159, in deserialize_library
09:38:00      tool = self.deserialize_tool(name)
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\serializers\fcserializer.py", line 286, in deserialize_tool
09:38:00      param, value = tool_property_to_param(propname, value, prop)
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\fcutil.py", line 84, in tool_property_to_param
09:38:00      value = parse_distance(value)
09:38:00    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\fcutil.py", line 62, in parse_distance
09:38:00      raise NotImplemented('unsupported value in file: "{}"'.format(value))
09:38:00  TypeError: 'NotImplementedType' object is not callable

ShamanTcler, 2023-07-25:
Installed via the add-on manager instructions (pulling from Github)

Version 0.9.0

knipknap, 2023-07-25:
Thanks for the bug report! Could you upload your tool library for me to test?

knipknap, 2023-07-25:
(If you can not, then you could update BTL and try again - this time it should generate a better error message)

knipknap, 2023-07-25:
Actually, there is a chance that this now already works... but BTL will convert any distance units to mm. I'll have to look into updating it to support other stuff.

ShamanTcler, 2023-07-25:
Hmm ... upload my tool library?  I looked in C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library, and yup I see your stuff there.

After the update you just made I get:
11:52:42  Better Tool Library loaded successfully.
11:52:42  Feeds and Speeds Addon loaded: PathWorkbench

 Library path is C:/Users/cesla/AppData/Roaming/FreeCAD/ToolBits

11:52:48  Warning: Skipping loading of the following files, because they have reserved names used by a builtin shape: ballend.fcstd, bullnose.fcstd, chamfer.fcstd, drill.fcstd, endmill.fcstd, probe.fcstd, slittingsaw.fcstd, thread-mill.fcstd, v-bit.fcstd --> To remove this warning, delete or rename the files
11:52:48  Traceback (most recent call last):
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\ui\init.py", line 22, in on_library_open_clicked
11:52:48      dialog = LibraryUI(tool_db, serializer)
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\ui\library.py", line 51, in __init__
11:52:48      self.load()
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\ui\library.py", line 101, in load
11:52:48      self.tooldb.deserialize(self.serializer)
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\db.py", line 114, in deserialize
11:52:48      self.deserialize_libraries(serializer)
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\db.py", line 84, in deserialize_libraries
11:52:48      for library in serializer.deserialize_libraries():
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\serializers\fcserializer.py", line 123, in deserialize_libraries
11:52:48      return [self.deserialize_library(id)
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\serializers\fcserializer.py", line 123, in <listcomp>
11:52:48      return [self.deserialize_library(id)
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\serializers\fcserializer.py", line 159, in deserialize_library
11:52:48      tool = self.deserialize_tool(name)
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\serializers\fcserializer.py", line 286, in deserialize_tool
11:52:48      param, value = tool_property_to_param(propname, value, prop)
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\fcutil.py", line 95, in tool_property_to_param
11:52:48      value = parse_distance(value)
11:52:48    File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\better-tool-library\.\btl\fcutil.py", line 73, in parse_distance
11:52:48      raise NotImplementedError('me value in file: "{}"'.format(value))
11:52:48  NotImplementedError: me value in file: "0.5"

Looking at some lines prior to line 22 in init.py. I uncommented  line 15 to get:

**Library path is C:/Users/cesla/AppData/Roaming/FreeCAD/ToolBits**

Going to that dir I find my file Whiteside.fctl (it was included as a txt file but exists as an fctl file)

I sure hope my earlier work is not breaking yours.

[Whiteside.txt](https://github.com/knipknap/better-tool-library/files/12163430/Whiteside.txt)

knipknap, 2023-07-25:
If you want me to test with your library, you will have to send me your entire tool directory including the bits and shapes.

However, I just completely refactored parameter handling to tolerate non-metric units, as I suspect that was the cause of your traceback. In other words, if you upgrade and try again, it should in theory work now.

That said: Make sure to backup your tool dir before trying!

ShamanTcler, 2023-07-25:
Good news .... it understands Whiteside bits ..

You can close this issue and I will play with it a bit more

Regards
Carl

[image]
[image]

knipknap, 2023-07-25:
That's great to hear, thanks for reporting the issue!

closed 2023-07-25 by knipknap