knipknap/better-tool-library #3: Crash report

ShamanTcler, 2023-08-08:
Win 11
FreeCad ver 0.21.0 (build 33682)
Better Tool Lib ( no version number, but "up-to-date" 8/8/2023

Tool displays fine.
Go to Feeds & Speeds
Added a machine: OneFinity
set some resonable machine values, returned ..

then crash,

Reloaded....

Feeds & Speeds loads with a  bunch of red/orange issues.

Red:
     Axial Force 0.00 N  ...min/max range given
     Mrr= 0.00 min/max range given
     Power= 2.2 Kw ...... **This is in fact what I put in ... limit was 2.2 max**
     Torque= 0.00
     
  

Orange:
    Deflection 0.02   ,im/max given
     RPM= 59188 .... I don't recall entering this

So it appears to be a matter of setting reasonable defaults.

Also it was not clear how to add/edit machines except through the tool editor.

ShamanTcler, 2023-08-08:
[image]

[image]

knipknap, 2023-08-08:
Those results look wrong indeed. Could you show your tool parameters?

Given the tiny stickout of just 3.75mm... this looks like the tool would be like 0.something mm thick? That would lead to extremely low forces and may explain at least part of the result, though I guess it should still have found at least something.

ShamanTcler, 2023-08-08:
This was a "kick the tires" exercise. So pretty much took defaults when able.
[image]

[image]

Note all my dims are in inches ... might that be an issues?

knipknap, 2023-08-08:
Ohh, I see... unfortunately, imperial units are not supported by the calculator; this is a documented limitation, see [here](https://github.com/knipknap/better-tool-library/blob/main/docs/feeds-and-speeds.md). I am sure that is what causes the issues.

I should probably detect if there is anything imperial going on and print an error instead of letting the calculator fail.

ShamanTcler, 2023-08-09:
I have not looked at you code, but what I have done in the past is:

1) Copy all parameters to temporary working variables
2) Check units, if need be, Set a flag, change working variables
3) Do the calcs (only need to validate once)
4) Check Flag, if set change units on returning vars
6) Return variables

I know simple, but simple can be good

ShamanTcler, 2023-08-09:
So I glanced at you code, well done, however some comment are best left out:

[image]

knipknap, 2023-08-11:
Sure, this requires just adding support in the presentation layer. In the calculator I even made things extra easy: All of the calculator's return values have a [get_imperial()](https://github.com/knipknap/better-tool-library/blob/1c62ffc8894a26d142524f0e3feeeea61c22bb0b/btl/feeds/param.py#L22) method that apply the correct factors for you.

So all that needs to be done is
- Changing the [machine editor UI](https://github.com/knipknap/better-tool-library/blob/1c62ffc8894a26d142524f0e3feeeea61c22bb0b/btl/ui/machineeditor.py) to convert the input/output as needed.
- Making the [tool property editor](https://github.com/knipknap/better-tool-library/blob/main/btl/ui/feedsandspeeds.py) UI type aware.
- Displaying the above mentioned result values in the [Feeds & Speeds UI](https://github.com/knipknap/better-tool-library/blob/main/btl/ui/feedsandspeeds.py) if the FreeCAD preferences are set to imperial.

Probably relatively straight forward, but it does require some effort.

knipknap, 2023-08-25:
Current master now supports imperial in all input fields, and serializes the units as well.

The calculator output is still metric, but that is a different matter. At least the calculator should no longer fail.

knipknap, 2023-08-26:
Closing this issue as the original problem should be resolved. If the original problem reappears, please feel free to reopen.

As for imperial output, that should be a new issue, and I do accept patches, but am not planning on working on it myself.

closed 2023-08-26 by knipknap

ShamanTcler, 2023-08-26:
Sounds good and thanks