knipknap/better-tool-library #4: exception in tool.py

spanner888, 2023-08-13:
In tool.py line 325, if shank_l == non_cutting, then an exception can occur:  ZeroDivisionError: float division by zero

```
#325        return min((yield_strength*solid_inertia) / ((shank_d/2)* shank_l),
                   (yield_strength*fluted_inertia) / ((diameter/2) * (shank_l+non_cutting)))
```

Suggested fix is below, but I am uncertain if this is correct understanding.
`#323        non_cutting = abs(cutting_edge-doc)`

knipknap, 2023-08-13:
Is this in the latest version? I recently fixed several division by zero issues, and the line numbers you included no longer match with the current main branch, so perhaps this issue is already fixed.

spanner888, 2023-08-13:
You are correct, I forgot to update to latest code, however the potential issue is still present:
`#339  non_cutting = cutting_edge-doc`

knipknap, 2023-08-13:
Huh, "cutting_edge-doc" should already be guaranteed to be >0... but I guess an additional safehold doesn't hurt. I committed the change. Feel free to reopen if the problem persists.

closed 2023-08-13 by knipknap