knipknap/better-tool-library #5: No module named 'scipy'

Schildkroet, 2023-08-19:
Hello,

i installed current version (0.9.5) via Addon-Manager.
Now i get following error:
During initialization the error "No module named 'scipy'" occurred in `/home/user/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/btl/./InitGui.py`

knipknap, 2023-08-19:
That's... odd. I think scipy is needed by FreeCAD, so it seems like your FreeCAD installation is incomplete.

You'll have to install scipy (pip install scipy) somewhere where FreeCAD finds it

Schildkroet, 2023-08-19:
I use flatpak version. Therefore i assume everything should be installed correct

knipknap, 2023-08-19:
Then the flatpak is missing scipy.

I added it to the BTL requirements file now, even though I think the better fix is having the flatpak ship with scipy. But it could be a workaround for the time being.

closed 2023-08-19 by knipknap

Schildkroet, 2023-08-19:
Now i get following:
```
/home/user/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/AdditionalPythonPackages/py310/scipy/__init__.py:132: UserWarning: A NumPy version >=1.21.6 and <1.28.0 is required for this version of SciPy (detected version 1.21.4)
  warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"
```
But the addon at least works.

knipknap, 2023-08-19:
Yeah, that is why a real fix should IMO be done in the flatpak, because BTL has no way of knowing in which context you are installing it, i.e. which version of scipy is compatible.

You won't be able to use the feeds&speeds calculator without a working scipy.

Schildkroet, 2023-08-19:
The addon seems to work as expected now.

knipknap, 2023-08-19:
Awesome 👍