knipknap/better-tool-library #9: How to increase resolution of tool pictograms?

ModuloFS, 2023-10-22:
Hello,

i am using a 4k monitor and if i open the better-tool-library application, the pictograms of the tools have a very low resolution as you can see in the attached picture. Is it possible to change this behavior to get better resolution?
[image]

knipknap, 2023-10-22:
Hey, thanks for the report - I don't have access to my PC for the rest of the year unfortunately, so I am afraid I can't provide a short term fix if it requires a code change.

Is the image you are using an SVG? Does the problem exist for built-in shapes as well, or just your own? From the screenshot it looks like it may be your tool only, indicating a DPI setting within the SVG file, perhaps the default size is set incorrectly in the file?

If you are willing and if it is an SVG, I could also ship it as a built in shape with BTL in the future.

ModuloFS, 2023-10-22:
It appears on build in tools as you can see on tool 6 and 7 as well as on self made ones like the tool 16. I have to say the self made one is a .png image. But it has also a resolution of 712x1369 and a dpi value of 96 as your included svg files. 
Is there a place in you code on wich the rendered resolution is changeable?

knipknap, 2023-10-22:
I see. Well, it would have to be in the renderer then, this is a very small piece of code:

https://github.com/knipknap/better-tool-library/blob/main/btl/ui/shapewidget.py

Or, where it is loaded:

https://github.com/knipknap/better-tool-library/blob/main/btl/ui/util.py

But I am not sure how to best do it such that it works independent of the screen size.

ModuloFS, 2023-10-22:
Ok, if i change QtCore.QSize(200, 235) in the shapewidget.py to QtCore.QSize(400, 470) the icon in the tools parameter editor window gets bigger but the dpi resoluion stays roughly the same. The preview of the tools as you can see in my picture from the beginning of the post stays exactly the same. The resolution does not change. I have tried to change the dpi value of my png file to 192 but that chaned nothing.

knipknap, 2024-02-21:
Ok, I think I was able to fix it (for SVG icons only). Just pushed the change, can you confirm that it works for SVG icons now?

knipknap, 2024-02-24:
Alright, it seems that this is a general problem of Qt, at least in PySide with Qt 5, which is what FreeCAD uses. I found [someone showing the same problem in a minimal example](https://github.com/mliberty1/pyside2_high_dpi/). While he found a workaround by using a stylesheet to load an SVG instead of using a Pixmap, this cannot work in BTL, because BTL stores the SVG in-memory, where it cannot be referenced by a stylesheet.

So unfortunately, it looks like the remaining issue will have to be fixed in Qt or PySide and I cannot do much about it - at least I found no other workaround that could work.

luzpaz, 2024-12-24:
Has there been any progress within qt6/pyside6 that impacts this issue ?