knipknap/better-tool-library #11: Tapered ball nose endmills

sgofferj, 2023-12-22:
How would I go about to add a tapered ball nose end mill? I don't see any option. If there is, please slam me with an "RTFM". If there isn't please consider it a feature request :).

knipknap, 2023-12-23:
There currently aren't any instructions yet, but the short story is:

- Use one of the existing shape files as a template to create your own shape: https://github.com/knipknap/better-tool-library/tree/main/btl/resources/shapes
- To make the shape dimension sketch work, also provide the SVG file, making sure that in the SVG, any text items have an attribute that corresponds with the attribute names in the FcStd file. This can be done in Inkscape using the UI, or by directly editing the SVG as a text file.

As I am traveling, I can't currently make screenshots to explain this better.
With both files ready, just place them in the shape directory in your tool library, and BTL should find it.

knipknap, 2024-02-20:
I added [documentation here](https://github.com/knipknap/better-tool-library/blob/main/docs/shape.md). Please let me know if I can improve anything more.

closed 2024-02-20 by knipknap