knipknap/better-tool-library #14: Binging up imperial output again ...

ShamanTcler, 2024-01-14:
I have started looking into this again. I am attempting to make a few changes, let's see if I am headed the right direction.  ( I am a C++ kind guy with some Python... so I know enough to be dangerous)

I see in btl\ui\feedsandspeeds.py a method _on_calculator_finished, it appears to be the routine that cycles through the parameters (param) and send them into the screen (table) pipeline. 

The inputs to the calculator can be metric or imperial, but the calculator always works in metric.  Which means the state of a parameter will always be metric.

My thoughts are to modify "params" to have a method "fmtd_to_display". If the flag mentioned earlier is set, convert from metric to imperial.

So now in _on_calculator_finished you call params.fmtd_to_display in the for row, (name, param) in enumerate(params): loop.

knipknap, 2024-01-14:
My thoughts: I would consider the params a backend function which should be unaware of the UI - so I would choose one of the following approaches:

- Option 1: `fmtd_to_display()` should be a `btl/ui/util.py` function (i.e. not a method) that can be invoked in the UI when the value is prepared for display.
- Option 2: The backend should have a method like `as_unit(theformat)`, where the format is the desired target format.

Your proposal is pretty much option 2, if it ensures that the param object does not access any UI specific stuff (i.e. it should make any calls to Qt functions). The method should create the target format based on the input parameters only.

A challenge may be that the UI does not necessarily know the target format. E.g. what is the target unit for a param that uses N(ewton) be? But `btl/units.py` has a function `get_default_unit_conversion()` that may be of help.

ShamanTcler, 2024-01-14:
Regarding your comment "the UI does not necessarily know the target format."  Would it not make sense that the units from the "Tool" tab match the units in the "Feeds & Speeds" tab?

knipknap, 2024-01-14:
I think not - especially when thinking about "shareable tools", but also when considering tools with "mixed specifications" - some dimensions may be specified in metric, others in inches.
Imagine that you import a tool for which all data was given in mm by the manufacturer - I am guessing you probably still want to see the calculator result in inches, or feet/minute, etc.?

In other words, I would display it always according to user preferences, regardless of what the specs of the tool were.

But maybe the user preference is "use the same unit as whatever the manufacturer specified", then that would be another reason to manage this complexity in the UI instead of the backend.

ShamanTcler, 2024-01-14:
First let me say that I view this tool as a tremendous asset, so please don't take my comments as being negative. I have a OneFinity Elite Foreman ( 2.2kw spindle with ATC, with a Masso G3 touch controller)

Running through a few ideas on displayed units:

Given: "I would display it always according to user preferences, regardless of what the specs of the tool were." does this mean you have a unit toggle under the preferences or is it on the actual "feeds and speeds" tab.  Actually, this could be convenient.... imagine typing in the parameters in imperial hitting OK and have them auto-magically converted to metric.

Or should the app display results (and input) as governed by: Edit/Preferences/General ... in my case Imperial decimal vs "Metric small parts & CNC"

knipknap, 2024-01-14:
Thanks, I am glad to see that the work I put in is useful ;-).

Since FreeCAD already has user settings for preferred units, my instinct would be:

1. Test whether these settings can be accessed when the FreeCAD GUI isn't up. In other words, when BTL is running in standalone mode, i.e. when starting it outside of FreeCAD. If this works, it is the way to go.
2. If this doesn't work, then I would put the setting into the BTL preferences. It may then make sense to have those preferences default to whatever is configured in the FreeCAD settings when the FreeCAD GUI is up.

Connor9220, 2024-04-24:
So, on the Feeds and Speeds output, it needs to follow the FreeCAD's units.  Also, no one uses mm/min  It's mm/sec and for imperial it's either ipm (inches per minute)  or in/min notation.