knipknap/better-tool-library #16: Main screen hard to read when using Dark Theme

Connor9220, 2024-01-16:
[image]

Sub pages look a bit better

[image]

All end mill images could use more contrast or something.

knipknap, 2024-01-16:
I added some logic to improve contrast in dark themes, but it seems that the ProDark theme is simply broken and overwrites this. In other words, it styles things that just shouldn't be styled - even though I explicitly assigned the text color to each of the labels, the color is still not used when that theme is active.

It looks like this cannot be fixed in BTL - or if anyone knows how please provide a patch.

Connor9220, 2024-01-16:
Here's what the main screen looks like in OpenDark

[image]

knipknap, 2024-10-13:
I now understand theming of FreeCAD better than when the issue was opened: This is actually not something BTL can or should control.

In other words, such bugs need to be reported to the theme author instead - I cannot fix it.

closed 2024-10-13 by knipknap