knipknap/better-tool-library #20: Icon not displayed

DavidHenri008, 2024-01-26:
I just installed the latest 0.9.9 version in my FreeCad 0.21.2 and unfortunately, the icon is not displayed.
I rebooted several times and still not showing...

[image]
[image]

knipknap, 2024-01-26:
Looks like you are using a FreeCAD with an old Python version. The error references a line that uses the new "match" statement, which is available starting from Python 2.10.

DavidHenri008, 2024-01-26:
I have python version 3.10 installed.
[image]
But I am not pointing to it in the settings
[image]

knipknap, 2024-01-26:
I am not sure how that works on Windows. Perhaps FreeCAD ships with an included Python? In any case, does it work when you define the path?

DavidHenri008, 2024-01-26:
Unfortunately not. I still have the issue after I set the path.

DavidHenri008, 2024-01-26:
Python 3 is not compatible with python 2. Do you run this with version 2 or 3 of python?

knipknap, 2024-01-26:
Python 2 is EOL. BTL uses Py 3 only.

knipknap, 2024-01-26:
I just pushed a commit to remove the match statement. It should now be compatible with Python 3.9

DavidHenri008, 2024-01-26:
I will give it a try.
It is strange because the match statement was supported from 3.10 and up... 
I will try to identify which version of python is executed by my FreeCad.

DavidHenri008, 2024-01-26:
Oh wow... look at this, I am somehow running version 3.8:
[image]

DavidHenri008, 2024-01-26:
So on windows, the python version is distributed with FreeCad...

knipknap, 2024-01-26:
It shouldn't matter anymore, BTL master should work anyway.

knipknap, 2024-02-08:
As the compatibility issue was fixed, I am closing the issue. Feel free to re-open if the problem persists.