knipknap/better-tool-library #25: Wrong numpy version on Freecad 0.21 realthunder branch flatpak

catgarlicc, 2024-02-10:
While loading the addon FreeCAD tells that the numpy version is 1.21 and that the addon is unable to initialize

```noracarmi@ilo-nanpa ~> flatpak run org.freecadweb.FreeCAD.Link
FreeCAD 0.21.0, Libs: 2024.123.0R14555 (Git shallow)
Gtk-Message: 13:50:42.273: Failed to load module "canberra-gtk-module"
Gtk-Message: 13:50:42.273: Failed to load module "pk-gtk-module"
Gtk-Message: 13:50:42.273: Failed to load module "canberra-gtk-module"
Gtk-Message: 13:50:42.273: Failed to load module "pk-gtk-module"
Could not find color scheme  ""
Qt: Session management error: Could not open network socket
/app/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py:142: UserWarning: A NumPy version >=1.22.4 and <1.29.0 is required for this version of SciPy (detected version 1.21.4)
  return original_import(name, *args, **kwargs)
RuntimeError: module compiled against API version 0xf but this version of numpy is 0xe
During initialization the error "numpy.core.multiarray failed to import" occurred in /home/noracarmi/.var/app/org.freecadweb.FreeCAD.Link/data/FreeCAD/Mod/btl/./InitGui.py
Please look into the log file for further information
A3 available
```

typing `pip list` gives me `numpy 1.21.4` as well as a bunch of error messages : 
```--- Logging error ---
Traceback (most recent call last):
  File "/app/lib/python3.10/site-packages/pip/_internal/utils/logging.py", line 177, in emit
    self.console.print(renderable, overflow="ignore", crop=False, style=style)
  File "/app/lib/python3.10/site-packages/pip/_vendor/rich/console.py", line 1673, in print
    extend(render(renderable, render_options))
  File "/app/lib/python3.10/site-packages/pip/_vendor/rich/console.py", line 1305, in render
    for render_output in iter_render:
  File "/app/lib/python3.10/site-packages/pip/_internal/utils/logging.py", line 134, in __rich_console__
    for line in lines:
  File "/app/lib/python3.10/site-packages/pip/_vendor/rich/segment.py", line 249, in split_lines
    for segment in segments:
  File "/app/lib/python3.10/site-packages/pip/_vendor/rich/console.py", line 1283, in render
    renderable = rich_cast(renderable)
  File "/app/lib/python3.10/site-packages/pip/_vendor/rich/protocol.py", line 36, in rich_cast
    renderable = cast_method()
  File "/app/lib/python3.10/site-packages/pip/_internal/self_outdated_check.py", line 130, in __rich__
    pip_cmd = get_best_invocation_for_this_pip()
  File "/app/lib/python3.10/site-packages/pip/_internal/utils/entrypoints.py", line 58, in get_best_invocation_for_this_pip
    if found_executable and os.path.samefile(
  File "/usr/lib/python3.10/genericpath.py", line 101, in samefile
    s2 = os.stat(f2)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/bin/pip'
Call stack:
  File "/app/bin/pip", line 8, in <module>
    sys.exit(main())
  File "/app/lib/python3.10/site-packages/pip/_internal/cli/main.py", line 70, in main
    return command.main(cmd_args)
  File "/app/lib/python3.10/site-packages/pip/_internal/cli/base_command.py", line 101, in main
    return self._main(args)
  File "/app/lib/python3.10/site-packages/pip/_internal/cli/base_command.py", line 223, in _main
    self.handle_pip_version_check(options)
  File "/app/lib/python3.10/site-packages/pip/_internal/cli/req_command.py", line 190, in handle_pip_version_check
    pip_self_version_check(session, options)
  File "/app/lib/python3.10/site-packages/pip/_internal/self_outdated_check.py", line 236, in pip_self_version_check
    logger.warning("[present-rich] %s", upgrade_prompt)
  File "/usr/lib/python3.10/logging/__init__.py", line 1489, in warning
    self._log(WARNING, msg, args, **kwargs)
  File "/usr/lib/python3.10/logging/__init__.py", line 1624, in _log
    self.handle(record)
  File "/usr/lib/python3.10/logging/__init__.py", line 1634, in handle
    self.callHandlers(record)
  File "/usr/lib/python3.10/logging/__init__.py", line 1696, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.10/logging/__init__.py", line 968, in handle
    self.emit(record)
  File "/app/lib/python3.10/site-packages/pip/_internal/utils/logging.py", line 179, in emit
    self.handleError(record)
Message: '[present-rich] %s'
Arguments: (UpgradePrompt(old='22.2.2', new='24.0'),)
```

What's super strange is that inside the addon manager FreeCAD tells me numpy's version is 1.26

[image]

I checked in the python console inside FreeCAD and it tells me it's running numpy 1.21 instead

[image]

What's the reason of this discrepancy ? There seems to be two distinct numpy versions coexistinp but the flatpak contained environment only sees numpy 1.21 and the addon manager only sees 1.26, and for some reason BTL tries to run against numpy 1.21. 

I'm kinda lost on this :(

(I'm running FreeCAD 0.21 realthunder's branch Stable 2024.01.23 release flatpak on Fedora 38)

Thank you in advance and thank you for the addon

knipknap, 2024-02-10:
Yep, that seems to be a bug in the appimage unfortunately. I reported it [here](https://forum.freecad.org/viewtopic.php?p=699304#p699304) a while ago, and it appears fixed in their current weekly development snapshot version.  I'm running that one myself, more specifically:

```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35759 (Git) AppImage
Build type: Release
Branch: main
Hash: 90d36f60c30cb6549a7987fcd9d4769eab1c160c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
```

This works fine.

knipknap, 2024-02-10:
Great to hear! Thanks for the reporting the issue!

closed 2024-02-10 by knipknap