knipknap/better-tool-library #27: Errors on opening

ShamanTcler, 2024-03-25:
Using DEV version as you have added "CAM WB" support

OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36526 (Git)
Build type: Release
Branch: main
Hash: 2ebbd83268ffabf41ceaad802714a4ffd9214437
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
  * Curves 0.6.29

Process: 
**Open** CAM WB 
10:51:29  Better Tool Library v0.9.9-53-g92fe1a0 loaded successfully.

**Click** BTL icon
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Endmill.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Endmill.fctb
10:52:38  Ouch! Unsupported attribute 'SpindleDirection' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Endmill.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Drill.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Drill.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Ball_End.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Ball_End.fctb
10:52:38  Copying required but non-existent shape: C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Shape\bullnose.fcstd
10:52:38  0 <App> Document.cpp(2147): Failed to restore _fusion_center_drill#Sketch.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
10:52:38  0.0219593 <App> Document.cpp(2147): Failed to restore _fusion_chamfer_mill#Sketch.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Bullnose.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Bullnose.fctb
10:52:38  Ouch! Unsupported attribute 'SpindleDirection' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Bullnose.fctb
10:52:38  Copying required but non-existent shape: C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Shape\v-bit.fcstd
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\60degree_Vbit.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\60degree_Vbit.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\45degree_chamfer.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\45degree_chamfer.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\slittingsaw.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\slittingsaw.fctb
10:52:38  Ouch! Unsupported attribute 'SpindlePower' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\probe.fctb
10:52:38  Copying required but non-existent shape: C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Shape\thread-mill.fcstd
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm-thread-cutter.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm-thread-cutter.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\45degree_chamfer.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\45degree_chamfer.fctb
10:52:38  Copying required but non-existent shape: C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Shape\thread-mill.fcstd
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm-thread-cutter.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm-thread-cutter.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Drill.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Drill.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Endmill.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Endmill.fctb
10:52:38  Ouch! Unsupported attribute 'SpindleDirection' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\5mm_Endmill.fctb
10:52:38  Copying required but non-existent shape: C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Shape\v-bit.fcstd
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\60degree_Vbit.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\60degree_Vbit.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Ball_End.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Ball_End.fctb
10:52:38  Copying required but non-existent shape: C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Shape\bullnose.fcstd
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Bullnose.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Bullnose.fctb
10:52:38  Ouch! Unsupported attribute 'SpindleDirection' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\6mm_Bullnose.fctb
10:52:38  Ouch! Unsupported attribute 'SpindlePower' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\probe.fctb
10:52:38  Ouch! Unsupported attribute 'Flutes' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\slittingsaw.fctb
10:52:38  Ouch! Unsupported attribute 'Material' with value 'None' in C:/Users/cesla/AppData/Roaming/FreeCAD/Macro\Bit\slittingsaw.fctb

knipknap, 2024-03-25:
Confirming, but still blocked by https://github.com/FreeCAD/FreeCAD/issues/13043 before I can resolve this...

knipknap, 2024-04-04:
Since the above FreeCAD issue was resolved I was able to try with the latest AppImage:

OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git) AppImage
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.60.0
  * btl 0.9.9
  * WebTools
  * toSketch
  * fasteners 0.4.62
  * FreeCAD_assembly3 0.11.0
  * lattice2 1.0.0
  * CycloidGearBox
  * fcgear 1.0.0
  * Assembly4 0.12.3

For me it loads without issues, so I suspect there was another FreeCAD issue causing the problem. Can you confirm whether it works with the latest dev branch of FreeCAD? (Of course you also need the latest version of BTL).

ShamanTcler, 2024-04-05:
yup seems to have cleared up