knipknap/better-tool-library #28: Multiple BTL icons on FC taskbar

napwodz, 2024-04-29:
`OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git) AppImage
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * btl 0.9.9
`

Whenever I change deactivate FreeCAD (switch windows, move to terminal or switch to web browser) I get another icon of BTL  on the taskbar:
[image]
[image]

And I have to deactivate every icon with a right click at the taskbar, unless I want a hundred of them after a while ;)
[image]

I know 0.22 is a dev version so far, but maybe someone knows the answer to my problem :)

knipknap, 2024-04-29:
I cannot reproduce it, but just added a safehold that should prevent it from happening anyway. If you update BTL I hope the problem is fixed.

napwodz, 2024-04-29:
> I cannot reproduce it, but just added a safehold that should prevent it from happening anyway. If you update BTL I hope the problem is fixed.

The problem is fixed - thank you! :)