knipknap/better-tool-library #29: Feed and spped not work

rmaltais, 2024-05-15:
I can't use this feature. When i select DOC and WOC always stay in certain percentage... Nothing happend after long time.

[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/Canada (fr_CA)
Installed mods: 
  * btl 0.9.9
[/code]

knipknap, 2024-05-16:
I'm afraid your report does not include enough information to find out what may be going on. Could you post a screenshot of the tool parameters, and another of the machine settings, and one of the calculator?

rmaltais, 2024-05-17:
Hi,
Sorry for not giving enough information. In fact, it is only in adatative mode that the calculator does not work. Here are the screenshots of the settings.

[image]
[image]
[image]

thank

knipknap, 2024-06-13:
Wow, looks like you had to dig pretty deep for this one, thank you! It's merged, so should work if you update.

Arm4g3ddon, 2024-06-13:
@knipknap Thanks man! Yeah a lot of back and forth but then I realized it was the method :)

closed 2024-06-14 by knipknap