knipknap/better-tool-library #32: when you create a new tool freecad closes

Vincenzoferrara, 2024-06-21:
https://github.com/knipknap/better-tool-library/assets/64591523/45844362-7765-4459-9cee-674d50d0c7ae

my version freecad is: 
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * AutoOpenSpreadsheet
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.5.16
  * Assembly4.backup1718498214.0742533 0.50.12 (Disabled)
  * fasteners 0.5.24
  * TabBar
  * Assembly4 0.50.13

knipknap, 2024-06-21:
Can you start FreeCAD from the command line and try the same again, to see if there is any command line output?

I suspect however, you'll see a segfault, which would indicate a problem with a library in the FreeCAD image you are using.

For reference, I am running successfully with the following AppImage:

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) AppImage
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.60.0
  * btl 0.9.9
  * WebTools
  * toSketch
  * fasteners 0.4.62
  * FreeCAD_assembly3 0.11.0
  * lattice2 1.0.0
  * CycloidGearBox
  * fcgear 1.0.0
  * Assembly4 0.12.3

```

knipknap, 2024-06-21:
I was able to reproduce it with the latest FreeCAD AppImage. And my suspicion was correct - it is a segfault in FreeCAD, not in BTL.

```
$ ./FreeCAD_weekly-builds-37819-conda-Linux-x86_64-py311.AppImage 
FreeCAD 0.22.0, Libs: 0.22.0devR37819 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.12.3) loaded
2.19163 <asm3.main> init_gui.py(14): no solver backend found
Cannot find icon: /tmp/.mount_FreeCAdUTbcc/usr/share/Mod/Web/Resources/icons/WebWorkbench.svg
Recent macros : keyboard shortcut(s) 1 disabled because of conflicts with Std_ViewFront respectively.
Hint: In Preferences -> Python -> Macro -> Recent Macros menu -> Keyboard Modifiers this should be Ctrl+Shift+ by default, if this is now blank then you should revert it back to Ctrl+Shift+ by pressing both keys at the same time.
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
Better Tool Library v0.9.9-61-g9c5b0ea loaded successfully.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f516be36520]
#1  0x7f5046980b10 in PartDesignGui::ViewProviderBody::unifyVisualProperty(App::Property const*) from /tmp/.mount_FreeCAdUTbcc/usr/lib/PartDesignGui.so+0x1d0
#2  0x7f5046986e9c in PartDesignGui::ViewProviderBody::onChanged(App::Property const*) from /tmp/.mount_FreeCAdUTbcc/usr/lib/PartDesignGui.so+0x3c
#3  0x7f5046490fcb in PartGui::ViewProviderPartExt::finishRestoring() from /tmp/.mount_FreeCAdUTbcc/usr/lib/./PartGui.so+0x2b
#4  0x7f516e8cd889 in Gui::Document::slotFinishRestoreObject(App::DocumentObject const&) from /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libFreeCADGui.so+0x49
#5  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libFreeCADApp.so(+0x2269ff) [0x7f516da049ff]
#6  0x7f516da1eb6c in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libFreeCADApp.so+0x52c
#7  0x7f516da1feed in App::Document::afterRestore(bool) from /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libFreeCADApp.so+0xbd
#8  0x7f516dc14b11 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libFreeCADApp.so+0x1c81
#9  0x7f516dc152f8 in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libFreeCADApp.so+0xc8
#10  0x7f516dc203c6 in App::Application::sOpenDocument(_object*, _object*, _object*) from /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libFreeCADApp.so+0x106
#11  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(+0x20603d) [0x7f516e00703d]
#12  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f516dfe6b5c]
#13  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f516dff2532]
#14  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x181) [0x7f516e017b71]
#15  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(+0x2f7d3f) [0x7f516e0f8d3f]
#16  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f516dfe6b5c]
#17  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f516dff2532]
#18  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x181) [0x7f516e017b71]
#19  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(+0x21e152) [0x7f516e01f152]
#20  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x24b) [0x7f516dfe6b3b]
#21  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f516dff2532]
#22  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x181) [0x7f516e017b71]
#23  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(+0x21e152) [0x7f516e01f152]
#24  /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x24b) [0x7f516dfe6b3b]
[... cut ...]
#87  0x7f516c50e212 in QCoreApplication::exec() from /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libQt5Core.so.5+0x82
#88  0x7f516e864e57 in Gui::Application::runApplication() from /tmp/.mount_FreeCAdUTbcc/usr/bin/../lib/libFreeCADGui.so+0xa27
#89  /tmp/.mount_FreeCAdUTbcc/usr/bin/freecad(+0x7620) [0x56427a1af620]
#90  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f516be1dd90]
#91  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f516be1de40]
#92  /tmp/.mount_FreeCAdUTbcc/usr/bin/freecad(+0x7981) [0x56427a1af981]
```

So it's not a BTL issue. You may open an issue on the AppImage instead.

closed 2024-06-21 by knipknap

Vincenzoferrara, 2024-06-23:
I installed the latest version now, the bug has been fixed