knipknap/better-tool-library #33: Cannot get BTL working in FreeCAD 0.22 (build 37841)

napwodz, 2024-06-23:
Hi.
I know 0.22 is a developer version and should not be used in manufacturing processes, but I found it very useful in many cases. Unfortunately BTL sopped working properly:
- when adding the package, the icon does not appear on the bar,
- when clicking TollTable (P,T) the standard taskbar appear on the right with Library Editor icon, but when clocking the latter one, BTL window pops out,
- cannot import any older library saved with BTL (in April 2024) - it is stacked with "unused tools", but when I try to add any tool from it, I get the following error:
`AttributeError: 'NoneType' object has no attribute 'get_tool_no_from_tool'`

OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * btl 0.9.9

Any idea what's going on?

knipknap, 2024-06-23:
I cannot reproduce any of the problems in the same AppImage with the latest BTL (main branch)

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.60.0
  * btl 0.9.9
  * WebTools
  * toSketch
  * fasteners 0.4.62
  * FreeCAD_assembly3 0.11.0
  * lattice2 1.0.0
  * CycloidGearBox
  * fcgear 1.0.0
  * Assembly4 0.12.3
```

When did you last update BTL?

napwodz, 2024-06-23:
The library was installed and updated today - just when I downloaded the latest AppImage of FC. Diretories .config/FreeCAD and .local/share/FreeCAD where freshly created of course ;)

knipknap, 2024-06-23:
Can you post the whole traceback.

knipknap, 2024-06-23:
Oh, can you check in the console what other output there is? Maybe there is this? "Unable to open Better Tool Library, because no writable tool directory was found."

napwodz, 2024-06-24:
Yep, you were right. I was mislead by previous experience - whenever I created the standard tool library (with FC creating new directories) I could not get the BTL working, so always, when hitting P,T, I was rejecting the query to create new directories for tool libraries and then I cout the BTL working. I don't know how that was possible, but that was the way I could use BTL. 

Now I had to do the opposite.

Thanks!