knipknap/better-tool-library #34: Tool Editor in FreeCad.

marblemonster, 2024-06-24:
Have 0.21 ...No Tool editor. Unable to alter existing tools. Wasn't able to import BTL
I have also not offset capabilities and the simulation cut is not complete.

Arm4g3ddon, 2024-06-25:
Hello marblemonster,

could you provide more Information? You can do this in FreeCad with Help -> About FreeCad -> Copy to Clipboard
Have you installed BTL via the Plugin Manager of FreeCad?

marblemonster, 2024-06-25:
Hi Tim,
Thanks for responding. I have tried to install BTL without success. My Tool
Dock does not respond with the create tool feature. I am sent to select a
shape and there it ends.
Even in Operations cant select any tools i tried to create.

Arm4g3ddon, 2024-06-25:
OK, have you created a Library first and selected it on the bottom left?

closed 2024-07-21 by knipknap

marblemonster, 2024-07-22:
Never got it to work. I'm a Fushion 360 guy, and was looking for a less
expensive software for my 3D printing.