knipknap/better-tool-library #35: Tool Parameter Editor will not resize vertically.

quick61, 2024-08-01:
For some reason the Tool Parameter Editor window will not resize vertically. The vertical size of the window seems to be hard coded and has no scroll bar and leaves no way to access whatever might be below the bottom horizon of the screen. I think there is other dialog (and maybe some buttons???) but have no way to know as the window can not be manipulated to view it. Even taking the window full size still leaves the bottom cut off.

knipknap, 2024-08-01:
I cannot reproduce it in master. Can you provide a screenshot, version info and resolution info, as well as other information that may be useful? (e.g. operating system, window manager)?

quick61, 2024-08-01:
Sure, sorry, should of included my FreeCAD About - 

OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * btl 0.9.9
  * CADExchanger
  * Curves 0.6.42
  * ExplodedAssembly
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0

Currently I am saddled with a monitor that can only do 720p rez. The main Tool Library window functions as it should, with both vertical and horizontal resizing and a scroll bar but 2 of those 3 window functions are missing from the Tool Parameter Editor window.

[image]

knipknap, 2024-08-01:
Scrollbar is now added in master.

closed 2024-08-01 by knipknap