knipknap/better-tool-library #37: Multiple icons of BTL

napwodz, 2024-09-10:
Hi again - I'm coming back with the same problem I had had a few months ago, but it was fixed very quickly ;)
Now, when I open FreeCAD (CAM framework active by default), I get two instances of BTL icons at the toolbar and whenever I change frameworks, I get an additional BTL icon every time I get back to CAM framework.

[image]
[image]

OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0

profknivre, 2024-09-11:
I've the same issue,

OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9

knipknap, 2024-09-15:
Thanks for reporting. It should be fixed; please reopen if the problem persists!

closed 2024-09-15 by knipknap

catgarlicc, 2024-11-21:
The dreaded icon invasion is happening again on 1.0.0 :(

In my case, it does not happen when switching workbenches, but whenever I refocus the FreeCAD window.

OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * btl 0.9.9
  * OpenTheme 2024.9.1

rmu75, 2024-11-28:
```
diff --git a/btl/ui/init.py b/btl/ui/init.py
index be903bd..5f2478a 100644
--- a/btl/ui/init.py
+++ b/btl/ui/init.py
@@ -81,16 +81,19 @@ def on_workbench_activated(workbench):
         _remove_toolbars(mw)
         return
 
-    # Create a toolbar if it does not yet exist.
-    toolbar = QtGui.QToolBar(mw)
-    toolbar.setObjectName(TOOLBAR_NAME)
-    mw.addToolBar(toolbar)
-
-    # Add the library editor button.
-    tool_button = QtGui.QToolButton(mw)
-    tool_button.setIcon(QtGui.QPixmap(ICON_FILE))
-    tool_button.clicked.connect(on_library_open_clicked)
-    toolbar.addWidget(tool_button)
+    toolbar = mw.findChild(QtGui.QToolBar, TOOLBAR_NAME)
+
+    if not toolbar:
+        # Create a toolbar if it does not yet exist.
+        toolbar = QtGui.QToolBar(mw)
+        toolbar.setObjectName(TOOLBAR_NAME)
+        mw.addToolBar(toolbar)
+
+        # Add the library editor button.
+        tool_button = QtGui.QToolButton(mw)
+        tool_button.setIcon(QtGui.QPixmap(ICON_FILE))
+        tool_button.clicked.connect(on_library_open_clicked)
+        toolbar.addWidget(tool_button)
 
     # Hack: Replace FreeCAD tool library by BTL by monkey-patching
     # the path workbench.
```

reneas, 2024-11-29:
I can reproduce it as well with alt+tabbing out of the window and in again
[image]

rmu75, 2024-11-29:
My diff kind of fixes the thing with additional toolbars appearing after alt-tabbing. I still get ```Better Tool Library v0.9.9-70-g2594eb9 loaded successfully.``` after each alt-tab or focus change.

knipknap, 2024-12-08:
The fix is pushed!

commit b1cbf9a7 (2024-09-15): try to fix #37 (multiple toolbar icons)