knipknap/better-tool-library #39: Default library export is not importable

smartin015, 2024-09-16:
* When I go to `File > Export Selected Library`, the file type defaults to `LinuxCNC .tbl (*.tbl)`
* When I go to `File > Import Selected Library`, the available options are `Camotics .ctbl (*.ctbl)`, `Fusion 360 .json (*.json)`, and `Fusion 360 .tools (*.tools)`

I run the appimage version of FreeCAD, which caused the tool libraries to be destroyed when the app terminated (due to it referencing a folder in /tmp/ with the unpacked appimage). I thought I could safeguard myself by exporting the library before restarting FreeCAD, but when I went to open it again I realized I'd exported to a non-importable format :)

It's not a huge issue, but it's proved a stumbling block to get a list of my own tools to persist past restarting FreeCAD. Would it be possible to add an import option for LinuxCNC, or at least default to the Camotics version since it's importable by BTL? Thanks!

knipknap, 2024-10-13:
If I understand you correctly, you want to use export/import to make a "backup" of existing tools.

However, this would be fundamentally flawed: Exports are inherently/unavoidably lossy, as third party tools do not support exactly the same data/attributes as FreeCAD. So no matter which export you use, you cannot re-import and expect it to work without losing information.

To backup, you would need to save the file as-ist - right now, there is no UI for that and you would have to copy your tool folder directly.

smartin015, 2024-10-14:
Thanks for spelling it out for me - I didn't realize there were export losses. Bummer.

I'll do as you recommend and just copy the whole folder - it'd be great to have a save option as part of the plugin, but I understand if it's not a priority :)