knipknap/better-tool-library #40: Tool tip and Name are not shown

pixelwaster, 2024-09-19:
Both the tool tip pop-up and status bar are not shown. The name is also not shown (check box is) when right click on the tool bar.

[image]

[image]

Possibly related is trying to use the wrong icon. The values are set in btl/ui/init.py (Line18) 
 In Report view:
```
11:33:44  Cannot find icon: Path_ToolTable
11:33:44  Better Tool Library None loaded successfully.
11:34:01  Better Tool Library None loaded successfully.
```
It should be CAM_ToolTable. Not really a python guy and I don't know enough to get the value of Pixmap but typing in the python console:
```
>>> 'CAMWorkbench' in FreeCADGui.listWorkbenches()
True
```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * btl 0.9.9
  * CAM
  * CurvedShapes 1.0.9
  * Curves 0.6.47
  * Draft
  * FeedsAndSpeeds 0.5.0
  * Fem
  * Help 1.0.3
  * Idf
  * Import
  * Inspection
  * lattice2 1.0.0
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot 2022.4.17
  * Points
  * ReverseEngineering
  * Robot
  * sheetmetal 0.4.26
  * Show
  * Silk 0.1.5
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web

```

knipknap, 2024-09-21:
I am confused; the line that you mention is already dynamic:

 'Pixmap': f'{wb}_ToolTable',

Are you sure that you are using the latest version of BTL?

pixelwaster, 2024-09-24:
I am using 0.9.9  even though  __version__ = NONE (set in  __init__.py )

HAS_CAM returns true in the python console
but in ( wb = 'CAM' if HAS_CAM else 'Path')  wb=='Path' and 'Pixmap': Path_ToolTable

MenuText is blank or null not "btl", "ToolBit Library editor"
ToolTip is blank as well and not  "btl", "Open an editor to manage ToolBit libraries"
TOOLBAR_NAME  != "btl_toolbar"

I don't know python. Just some crude poking about. I do know that init,py is run first and the values set here aren't showing up later. Using " 'CAMWorkbench' in FreeCADGui.listWorkbenches() " was to see if HAS_CAM might have returned false

knipknap, 2024-10-13:
Thanks for the report. I just pushed a fix. Please reopen if the problem persists in the latest version.

closed 2024-10-13 by knipknap