knipknap/better-tool-library #42: Only exports tool table in metric?

blazini36, 2024-10-15:
I generally use FreeCAD in metric units these days but I run LinuxCNC on the mill and it is setup for inches. Path (or CAM now) doesn't have a problem exporting in inches but this tool library doesn't seem to have any way to do it. I've specified tool diameters in inches, I've tried switching FreeCAD over to inches just to see if BTL was dependent on FreeCAD's units and do matter what it seems to convert all tool diameters to metric.

Should add some means to specify the output units.

knipknap, 2024-10-15:
Unfortunately there is no support for imperial output yet. See #14.

Any contributions would be gladly accepted

blazini36, 2024-10-16:
Well feeds and speeds are one thing, but the tooltable is actually being read by another machine. I'm not a programmer but I can understand the difficulty of adding imperial units to most of the functions in BTL but a small checkbox or something in the UI and converting only the diameters of each tool when exported seems like far less of a task. At least until real imperial support is added.

The way it is, every time you add a tool and export the tooltable, you manually have to convert every diameter. I'm sure if I knew what I was doing I could write a script to do it but unfortunately that's not the case. As bad as the built in tool library is, this one is basically not usable if the output is for an imperial machine for this simple reason.

knipknap, 2024-10-16:
I agree, that's why I left it open as a separate issue. Maybe it's more likely that someone adds the feature that way.