knipknap/better-tool-library #44: Include Tool holders in the tool configuration

AdrianCM95, 2024-11-06:
Hi,

I find this is a really useful project and here it is my proposal to make it even better. 

- **Proposal**: To include tool holders in the tool definition.
- **Reason**: The tool holder also needs to be consider for collisions during machining to avoid collisions. Additionally, it also helps to keep track of the other things like manufacturer or the part number. Most importantly, it looks really cool.
- **Implementation concept**:
+ To define a tool holder item: using a base model and changing parameters / using a 3D model / defining and stacking basic shapes based on a form.
+ To define an sub-library of tool holders.
+ To be able to add a tool holder to tool number "X" and use a parameter "tool stick-out" to define the distance between the tip of the tool and the tip of the tool holder.

This is an illustration of the concept:
[image]

Many tools, many tool holders, many beautiful combinations can be done ;)
[image]

knipknap, 2024-11-21:
Thanks for the well written proposal! Looks like it would really be a good addition, though for addressing the stickout I would assume that the CNC machine compensates for the tool stickout, without needing different gcode? But perhaps not all CNC machines do that.

AdrianCM95, 2024-11-21:
Hello @knipknap, thanks for the attention on this topic.

All the tool holder is not meant to modify the G-code, the tool is the reference. If you think of it as an assembly, the tool holder would have an offset constrain from the tip of the tool. The tool would be fixed.

When simulating the tool path, then you will be able to see if there is any collision betten the tool holder and the part. This is specially useful if you want to go deeper or if you have workholding around.

Let me know if this clarified your question.

knipknap, 2024-11-21:
Ah, I can see how it would be useful for the simulation. However, this is not maintained by BTL and would need changes in the CAM workbench to be possible.

AdrianCM95, 2024-11-22:
Okay, so this is only for cutting tools.

I don't think it makes sense to have this issue open here if it is not the right place. Here some options:
- Should it be moved to the FreeCAD issue list?
- Do you know any other project that is suitable for this request?
- Does it makes sense to include it in the scope of this project in the future or create a parallel project for this?

knipknap, 2024-11-22:
I think this issue does make sense here - BTL is a good place to store this kind of information. Only once the holders are documented, then it would make sense to use the information in the CAM processing. It's only this second step that would not be maintained in BTL.