knipknap/better-tool-library #54: BTL in Freecad 1.1 not working

RneeJ, 2025-03-04:
After about 2 years of not using Freecad i updated to version 1.1 .
I just started with the CAM Workbench and already made a design witch was milled (without the use of BTL).

I would like to use BTL .

Sadly it trows a lot of errors in Freecad version 1.0 and 1.1. :

```
13:13:23  Better Tool Library v0.9.9-81-g20271b2 loaded successfully.
13:13:30  Ouch! Unsupported attribute 'Flutes' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/5mm_Drill.fctb
13:13:30  Ouch! Unsupported attribute 'Material' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/5mm_Drill.fctb
13:13:30  Ouch! Unsupported attribute 'Flutes' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/6mm_Ball_End.fctb
13:13:30  Ouch! Unsupported attribute 'Material' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/6mm_Ball_End.fctb
13:13:30  Copying required but non-existent shape: /home/rene/Documents/a-Projecten/Kolomboormachine/Shape/bullnose.fcstd
13:13:30  1.5e-08 <App> Document.cpp(2420): Failed to restore _fusion_chamfer_mill#Sketch.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
13:13:30  0.0569601 <App> Document.cpp(2420): Failed to restore _fusion_center_drill#Sketch.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
13:13:30  Ouch! Unsupported attribute 'Flutes' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/6mm_Bullnose.fctb
13:13:30  Ouch! Unsupported attribute 'Material' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/6mm_Bullnose.fctb
13:13:30  Ouch! Unsupported attribute 'SpindleDirection' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/6mm_Bullnose.fctb
13:13:30  Copying required but non-existent shape: /home/rene/Documents/a-Projecten/Kolomboormachine/Shape/v-bit.fcstd
13:13:30  Ouch! Unsupported attribute 'Flutes' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/60degree_Vbit.fctb
13:13:30  Ouch! Unsupported attribute 'Material' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/60degree_Vbit.fctb
13:13:30  Ouch! Unsupported attribute 'Flutes' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/45degree_chamfer.fctb
13:13:30  Ouch! Unsupported attribute 'Material' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/45degree_chamfer.fctb
13:13:30  Ouch! Unsupported attribute 'Flutes' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/slittingsaw.fctb
13:13:30  Ouch! Unsupported attribute 'Material' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/slittingsaw.fctb
13:13:30  Ouch! Unsupported attribute 'SpindlePower' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/probe.fctb
13:13:30  Copying required but non-existent shape: /home/rene/Documents/a-Projecten/Kolomboormachine/Shape/thread-mill.fcstd
13:13:30  Ouch! Unsupported attribute 'Flutes' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/5mm-thread-cutter.fctb
13:13:30  Ouch! Unsupported attribute 'Material' with value 'None' in /home/rene/Documents/a-Projecten/Kolomboormachine/Bit/5mm-thread-cutter.fctb
13:13:30  Traceback (most recent call last):
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/ui/init.py", line 66, in on_library_open_clicked
13:13:30      dialog = LibraryUI(tool_db, serializer, parent=FreeCADGui.getMainWindow())
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/ui/library.py", line 80, in __init__
13:13:30      self.load()
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/ui/library.py", line 189, in load
13:13:30      self.db.deserialize(self.serializer)
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/db.py", line 143, in deserialize
13:13:30      self.deserialize_tools(serializer)
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/db.py", line 129, in deserialize_tools
13:13:30      for tool in serializer.deserialize_tools():
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/serializers/fcserializer.py", line 290, in deserialize_tools
13:13:30      return [self.deserialize_tool(id)
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/serializers/fcserializer.py", line 290, in <listcomp>
13:13:30      return [self.deserialize_tool(id)
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/serializers/fcserializer.py", line 346, in deserialize_tool
13:13:30      shape = self.deserialize_shape(shapename)
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/serializers/fcserializer.py", line 268, in deserialize_shape
13:13:30      shape = Shape(name, filename)
13:13:30    File "/home/rene/snap/freecad/common/Mod/btl/./btl/shape.py", line 122, in __init__
13:13:30      raise OSError('shape "{}" not found: {}'.format(name, self.filename))
13:13:30  OSError: shape "tap" not found: /home/rene/Documents/a-Projecten/Kolomboormachine/Shape/tap.fcstd
'''

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40505 (Git) Snap 1320
Build type: Release
Branch: main
Hash: b59cd794274c8c007b2ef16c3a06fecf30c0aef1
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/80.9812
Installed mods: 
  * btl 0.9.9

knipknap, 2025-03-05:
Could you upload your tool library somewhere by any chance? That would make debugging tons easier, I think

RneeJ, 2025-03-05:
These are the default tools that come with the default library in Freecad. I copied them to my project directory. maybe that is a problem.

RneeJ, 2025-03-06:
I un-re-installed BTL, I removed the default library items. Previous errors are gone now. Some others come up. But BTL is working now. Thanks.