knipknap/better-tool-library #55: New Release

Tectu, 2025-04-04:
Thank you for this great Add-On!

Would it be possible to get a new release?

knipknap, 2025-04-14:
Not likely. I am in the process of integrating BTL into FreeCAD. See https://github.com/FreeCAD/FreeCAD/issues/9466

snelweg, 2025-04-15:
> Not likely. I am in the process of integrating BTL into FreeCAD. See [FreeCAD/FreeCAD#9466](https://github.com/FreeCAD/FreeCAD/issues/9466)

So no more dumb whining about non translatable pictures from the FC people???? 
I was so disappointed to get that answer when I asked last year (!) why the integration did  not happen.
Hopefully the integration will be a smooth one this time, 
Thanks for all your work.